weliem/blessed-android-coroutines #3: Not showing in the screen

simpxlify, 2021-09-08:
Hello, I found out this by coincidende while studying bluetooth and coroutines on android studio kotlin and I found out your library in java which is amazing and works amazingly, I found out this one too just now and wanted to run a test if it worked as smooth as the java one but the text when you measure temperature is not showing, didnt test anything else but that for now but I dont know if you are going to look into this, in the logcat it all works wonderful.

weliem, 2021-09-08:
That is strange. Works totally fine on my phone.... can you try again or try on a different phone?

Make sure you have uninstalled the Java version if you have it installed....otherwise 2 apps are competing for the same data.

simpxlify, 2021-09-08:
I've tried with the heart rate machine and works fine, the temp one only works sometimes somehow, does it have to do with battery at all?
I'll test with a phone later for sure.

weliem, 2021-09-08:
As you can see in the code, all measurement types are treated the same way. Don't see anything different for the temperature measurement. Can you share a log for a case where it doesn't appear?

simpxlify, 2021-09-09:
The logs do appear, and after some tests came to conclusion that might be the thermometer fault because it only seems to send information when going off somehow, I was comparing the java version and on the temperature parte you actually used PNP_ID, would that change anything? Anyways might be device fault, need to find another one to test with.

simpxlify, 2021-09-09:
[image]
He gets the device but he cant read values or presenting them, sometimes he can, I just cant understand what is going on

martijnvanwelie, 2021-09-09:
As you can see in the logs, no temperature value was received. So nothing will be displayed.

simpxlify, 2021-09-09:
> As you can see in the logs, no temperature value was received. So nothing will be displayed.

Yes, but that's the thing, it's giving a temperature on the thermometer but it's just not getting passed, with some exceptions that sometimes they do pass occasionally, I also have another question if you could answer me, where did you get the uuid for all the ones u tried? I've searched on google and can't find for my pulse oximeter as an example.

weliem, 2021-09-10:
I see you are using the Taidoc 1241. It indeed behaves a bit strange. The first measurement normally comes through but the second doesn't. I usually disconnect 2 seconds after receiving measurements and then I wait for it to turn off (around 30 seconds). That sort of makes it manageable. But the real problem is the device itself...

simpxlify, 2021-09-10:
> I see you are using the Taidoc 1241. It indeed behaves a bit strange. The first measurement normally comes through but the second doesn't. I usually disconnect 2 seconds after receiving measurements [...]

Okay, so it is indeed the device himself, I'll try to get another thermometer with bluetooth, thank you very much sir.

simpxlify, 2021-09-10:
Alright I made them all connect, I have yet to get better understanding of this, Im doing a general scan for now because oximeter and scale werent getting connected, I get them all to connect but once again they just dont show their values or anything and I cant find the cause at all, the only one that always instantly show his values is the blood pressure one, I dont know what is failing here to them not to write on logcat or screen when they connect, if you have any suggestion please let me know, thank you in advance.

weliem, 2021-09-10:
Please share some logs, otherwise I have no clue what is going on....

simpxlify, 2021-09-10:
[image]
Im sorry, alright here are the logs for the pulse oximeter right as he starts I get values on the pulseOx but they are not passed to the mobile phone.

martijnvanwelie, 2021-09-10:
Ok, you should know that all Taidoc devices use a mix of standard services and proprietary services. The pulseoximeter you are using does not have the standard PulseOX service. You need to use the proprietary service and it's protocol of writing commands. You should contact Taidoc to get their official documentation.

simpxlify, 2021-09-10:
Okay, I will do that, sir I have one more question if you could answer, could you please explain me as an example this part of the code:

`val timestampPresent = flags and 0x02 > 0
val userIDPresent = flags and 0x04 > 0
val bmiAndHeightPresent = flags and 0x08 > 0`

I dont quite understood this flags and numbers, if you could please explain I would be grateful to you.

weliem, 2021-09-14:
The first byte is used to indicate whether certain content will be in the byte array. So every 'bit' of the 'byte' has a meaning. This code just tests which bits are on/off.

closed 2021-09-27 by simpxlify