weliem/blessed-android-coroutines #8: Send an array of bytes

simpxlify, 2021-11-18:
Hi once again, Im again using your library and I was thinking on a way of, for example I have a device lets say is TD4289 in this example and I want to send an array of bytes such as [0x51, 0x26, 0x00, 0x00, 0x00, 0x00, 0xA3, 0x1A] and read the information that comes with it, should be some data. Could I possibly do that? Send that exact array and await/read the data from the device?

As I dont fully understand bluetooth and BLE yet ff you could teach me how to achieve what I just said I would be really glad.

Much thanks.

weliem, 2021-11-18:
Github is not the place to get BLE education. I recommend you read a book on BLE....

But writing a byte array is not hard. In your case it would be something like this:

```kotlin
 val command = byteArrayOf(0x51, 0x26, 0x00, 0x00, 0x00, 0x00, 0xA3.toByte(), 0x1A)
 peripheral.writeCharacteristic(YOUR_SERVICE_UUID, YOUR_CHARACTERISTIC_UUID, command, WriteType.WITH_RESPONSE)
```

simpxlify, 2021-11-23:
I've messing around with it as I basically needed to learn brute forcefully about it to the company where I am right now as a junior, whatever, Ive done similar to the code you presented, and as I already could connect to it without sending the byte array but doesn't display info, so I was thinking, do I need a new approach to read when I send an array of bytes or not?

closed 2022-02-14 by weliem