weliem/blessed-android-coroutines #10: Mark location permissions with maxSdkVersion=30

nisrulz, 2021-11-23:
Location permissions are only required for API 30 and below. Marking those to be only for maxSdkVersion=30 should be part of the lib.

I already have a branch with the changes, but wanted to open an issue to discuss if there are any blockers or questions.

weliem, 2021-11-23:
Maybe it doesn't hurt. But why do you think it 'should be' part of the library? If you don't need it in your app you can just ignore it. What if somebody needs location permission for other reasons than bluetooth?

nisrulz, 2021-11-23:
It is already part of the library. Let me clarify:

These permissions are part of lib AndroidManifest file:
```xml
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"  />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
```
i.e during Manifest Merger phase these are merged into the final AndroidManifest.xml file along with that of the app. Now based on current setup, this would mean that the app is now listing itself to be using Location permissions for API 31+ when it actually doesn't need to because it only wants to use the Bluetooth functionality.

The core usecase of the lib is to provide Bluetooth functionality. Prior to API 31 the location permissions were required. Hence restricting the location permissions to API 30 and below will mean that they are not considered for API 31.

Note that you [already assert that location services](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#assert-never-for-location) are not to be used in the BLUETOOTH_SCAN permission by using `android:usesPermissionFlags="neverForLocation"`

```xml
<uses-permission
        android:name="android.permission.BLUETOOTH_SCAN"
        android:usesPermissionFlags="neverForLocation"
        tools:targetApi="s" />
```

weliem, 2021-11-25:
Ok feel free to raise a PR

closed 2021-11-26 by weliem

cren90, 2023-06-20:
@kwiky @nisrulz @weliem After fighting this for several hours before discovering the maxSdkVersion attribute on the permission in the Manifest Merge Blame. I am squarely in the camp of document the permissions needed and leave it out of the library. I know this is an old ticket so I'll open a new one if you guys agree

abdallahalaraby, 2023-10-16:
I just spent two days trying to figure out what's causing my app to not include the location permission when I already had added it to the app Manifest. The `tools:remove="android:maxSdkVersion"` MUST be added to the main Readme.md file

NikitaLeontev, 2024-05-16:
> @kwiky @nisrulz @weliem After fighting this for several hours before discovering the maxSdkVersion attribute on the permission in the Manifest Merge Blame. I am squarely in the camp of document the [...]

I spent 3 hours to solve the problem, please return my 3 hours drained into the trash

weliem, 2024-05-16:
This issue was already fixed a while ago. Also the Read.me was upated. 

Are you on the latest version of this library?

NikitaLeontev, 2024-05-17:
The application uses a location, after including the library in the build, when checking the location, the activity is recreated
Аfter the merge, the manifest contains a limit of 30 sdks, which breaks the normal operation of the application
@kwiky the solution works, but to understand what the problem is when adding a library, you need to spend time
https://github.com/weliem/blessed-android-coroutines/blob/main/app/src/main/AndroidManifest.xml (manifest has a limit of 30sdk for location)

nisrulz, 2024-05-17:
You are pointing to the permissions defined in the [sample app](https://github.com/weliem/blessed-android-coroutines/blob/main/app/src/main/AndroidManifest.xml) and not the [library module](https://github.com/weliem/blessed-android-coroutines/blob/main/blessed/src/main/AndroidManifest.xml). 

The one from the library module get merged into the YOUR app when you add a dependency on the library i.e

```xml
<!-- Needed only if code looks for Bluetooth devices.
         Since the code doesn't use Bluetooth scan results to derive physical
         location information, it is strongly asserted that the code
         doesn't derive physical location.
         Link:
     https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#assert-never-for-location
         -->
     <uses-permission
          android:name="android.permission.BLUETOOTH_SCAN"
          android:usesPermissionFlags="neverForLocation"
          tools:targetApi="s" />

     <!-- Needed only if code communicates with already-paired Bluetooth
         devices. -->
     <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

     <!-- Needed only if code makes the device discoverable to Bluetooth
         devices. -->
     <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
```
**Note that none of these have the `android:maxSdkVersion="30"` set on them. `tools:targetApi="s"` is for helping the IDE.**

The app module in this repo is for demonstrating the usage of the library module. The information to use the extra permissions is [in the readme of this repo](https://github.com/weliem/blessed-android-coroutines/tree/main?tab=readme-ov-file#adding-permissions). The documentation is actually clear. The reason you are having an issue is because you literally copy pasting the code in the Readme as it is rather than use logic about why you need the permission set in a certain way. 

At the very least the documentation can be updated to mention the 2 usecases i.e one where your app needs location permission as a feature + bluetooth and one where you only need it for bluetooth. 

I hope that clarifies how to use the library.

NikitaLeontev, 2024-05-17:
indeed, there is no such problem on the latest version, I had a branch written more than a year ago with version 0.3.2, and now when installing on a modern phone there was the problem described above

PR #11 (2021-11-26): Added maxSdkVersion to some permissions
Fixes #10