weliem/blessed-android-coroutines #14: Trying to get 16 byte from Characteristic but ouputs only zeros

AkroSportsBJ, 2022-02-21:
I'm using this code to read from a characteristic. It works in the NRF app as well as in some older code I used in Flutter (flutter blue library).

```
peripheral.getCharacteristic(SERVICE_UUID, NOTIFICATION_UUID)?.let {
   val test = peripheral.readCharacteristic(it)
   Timber.d("${test.size}")
   Timber.d("${test.asList()}")
}
```

As output, I only get zeros

```
/BluetoothPeripheral: reading characteristic <0000ff01-0000-1000-8000-00805f9b34fb>
D/BluetoothHandler: 16
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

That am I missing?

AkroSportsBJ, 2022-02-22:
I tried the BLESSED Java library as well now and it works. So there is a bug with this library. I'll hope we can find it and fix it as I would prefer to use this one :)

weliem, 2022-02-27:
The same code runs fine in the test app of this library. Looks like your peripheral is really returning 16 zeros...

AkroSportsBJ, 2022-02-28:
No, I also programmed the peripheral.

It works with the flutter_blue library on Dart and with the Java version of your library BLESSED as well as on nRF and Windows

martijnvanwelie, 2022-02-28:
The code for a readCharacteristic is very straightforward and since you are getting a byte array of the right length, I don't understand what could be going wrong. Especially since it works in my demo app...

Is there anything you can share to debug this issue? What's you code? Can you share a log?

closed 2022-04-16 by weliem