weliem/blessed-android-coroutines #17: Notify Listeners when Scan Fails

cren90, 2022-03-01:
Currently, the BluetoothCentralManager gets the failure status and logs it, but doesn't send it to listeners:
```         
    private fun sendScanFailed(scanFailure: ScanFailure) {
        currentCallback = null
        currentFilters = null
        scope.launch {
            Logger.e(TAG, "scan failed with error code %d (%s)", scanFailure.value, scanFailure)
         //   bluetoothCentralManagerCallback.onScanFailed(scanFailure)
        }
    }
```

Looks like it's just a matter of uncommenting line 124.

weliem, 2022-03-03:
It is a bit more complex than that. But I added a extra lambda parameter to get scan failures:

```
        central.scanForPeripheralsWithServices(supportedServices,
            { peripheral, scanResult ->
                Timber.i("Found peripheral '${peripheral.name}' with RSSI ${scanResult.rssi}")
                central.stopScan()
                connectPeripheral(peripheral)
            },
            { scanFailure -> Timber.e("scan failed with reason $scanFailure") })

```

weliem, 2022-03-03:
Released version 0.3.0

...let me know if this works for you

closed 2022-03-14 by weliem