weliem/blessed-android-coroutines #18: minSdk 26

FabioMatt83, 2022-04-12:
Hello Weliem, 
first of all, congrats on your SDK. My problem is that I'd like to import it into my app but since I need to support devices with at least Oreo on board (Api Level 24) it looks like I cannot because your SDK's minSdk is 24. Is there a reason for that?  Thanks

weliem, 2022-04-12:
Thanks! Well, I choose 26 because of the following reasons:

- Because that is the minimum for compiling to Java 8
- Because I could then drop <26 BLE bug workarounds and keep the code clean and simple
- We are at 31 now, so that gives 5 Android versions....

But then again, if you fork and modify the minSdk to 24 it will probably work...

FabioMatt83, 2022-04-12:
Thanks for your quick clarification!

closed 2022-04-16 by weliem