weliem/blessed-android-coroutines #20: BluetoothPeripheral GATT error NO_RESOURCES

imsunburnt, 2022-04-15:
E/BluetoothPeripheral: writing <e607040f08350e052c01> to characteristic <00002a2b-0000-1000-8000-00805f9b34fb> failed, status 'NO_RESOURCES'
E/BluetoothHandler$handlePeripheral: com.welie.blessed.GattException: GATT error NO_RESOURCES (128)
        at com.welie.blessed.BluetoothPeripheral$writeCharacteristic$4$result$1.onCharacteristicWrite(BluetoothPeripheral.kt:861)
        at com.welie.blessed.BluetoothPeripheral$bluetoothGattCallback$1$onCharacteristicWrite$1.invokeSuspend(BluetoothPeripheral.kt:201)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:920)

I am using Omron BP5450.  Running the example program.  The java version of the library works fine.  Phone is Samsung Galaxy A52 5G.

weliem, 2022-04-15:
That's not really a library issue. I probably forgot to deal with the gatt error in the example app...

imsunburnt, 2022-04-15:
Found issue: In BluetoothHandler "BLEsmart_" should be "BLESmart_" (at least for Omron BP5450).

                peripheral.getCharacteristic(CTS_SERVICE_UUID, CURRENT_TIME_CHARACTERISTIC_UUID)?.let {
                    // If it has the write property we write the current time
                    if (it.supportsWritingWithResponse()) {
                        // Write the current time unless it is an Omron device
                        if (!peripheral.name.contains("BLEsmart_")) {
                            val parser = BluetoothBytesParser(ByteOrder.LITTLE_ENDIAN)
                            parser.setCurrentTime(Calendar.getInstance())
                            peripheral.writeCharacteristic(it, parser.value, WriteType.WITH_RESPONSE)
                        }
                    }
                }

weliem, 2022-04-16:
Thanks for sorting this out! Committed a fix (bf12a2f5e47fa8055212c92b424c3bb012fc67d6)

imsunburnt, 2022-04-16:
Btw, brilliant piece of software!

closed 2022-05-03 by weliem