weliem/blessed-android-coroutines #21: "IllegalStateException: Already resumed" on disconnected internal callback

kwiky, 2022-05-13:
Hi

Using 0.3.0, we sometime have this exception

```
Fatal Exception: java.lang.IllegalStateException: Already resumed
       at kotlin.coroutines.SafeContinuation.resumeWith(SafeContinuationJvm.kt:44)
       at com.welie.blessed.BluetoothCentralManager$cancelConnection$2$1.onDisconnectedPeripheral(BluetoothCentralManager.kt:538)
       at com.welie.blessed.BluetoothCentralManager$internalCallback$1$disconnected$1.invokeSuspend(BluetoothCentralManager.kt:209)
       at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
       at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
       at kotlinx.coroutines.internal.LimitedDispatcher.run(LimitedDispatcher.kt:39)
       at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:95)
       at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
```

There is no way to catch it outise of the lib because the exception raise in an internal callback here : 
https://github.com/weliem/blessed-android-coroutines/blob/0.3.0/blessed/src/main/java/com/welie/blessed/BluetoothCentralManager.kt#L209

weliem, 2022-05-13:
Thanks for reporting. Not quite sure why this is happening.

How can I reproduce this?

weliem, 2022-05-13:
I guess you are seeing that the device is disconnected twice? Can you share more logs?

kwiky, 2022-06-10:
Yes, i think that my code call disconnect twice in some cases.
But i think that the lib have to support this behavior and manage it, by ignoring the second call or something like that 🤷🏻‍♂️ ?

closed 2025-02-07 by weliem