weliem/blessed-android-coroutines #29: use library on android wear 3 -> bonding failed

vkjambit, 2022-08-04:
so this is a special one ... i'm trying to use your library on an android wear 3 watch (samsung galaxy watch 4). 

on my phone your demo-app is asking me if i want to bond with my omron evolv and after pressing yes everything works. 
on my watch this popup is not shown and it immediately logs "bonding failed"

i don't think that you tried your library on a watch and especially not on wear os 3 but maybe you have an idea. 

log from my watch:

2022-08-04 15:48:54.924 15479-15479/com.example.bt_wear_spielwiese D/BluetoothPeripheral: starting bonding with 'BLESmart_0000021FEC21E500D0D6' (EC:21:E5:00:D0:D6)
2022-08-04 15:48:54.928 15479-16079/com.example.bt_wear_spielwiese I/BluetoothHandler$connectPeripheral: Bond state is BONDING
2022-08-04 15:48:54.938 15479-15479/com.example.bt_wear_spielwiese D/BluetoothPeripheral: pairing request received: PAIRING_VARIANT_CONSENT (3)
2022-08-04 15:48:55.028 15479-15479/com.example.bt_wear_spielwiese E/BluetoothPeripheral: bonding failed for 'BLESmart_0000021FEC21E500D0D6', disconnecting device      
2022-08-04 15:48:55.031 15479-16055/com.example.bt_wear_spielwiese D/BluetoothGatt: cancelOpen() - device: EC:21:E5:00:D0:D6
2022-08-04 15:48:55.031 15479-16079/com.example.bt_wear_spielwiese I/BluetoothHandler$connectPeripheral: Bond state is BONDING_FAILED       
2022-08-04 15:48:55.060 15479-16038/com.example.bt_wear_spielwiese D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=EC:21:E5:00:D0:D6
2022-08-04 15:48:55.083 15479-16038/com.example.bt_wear_spielwiese D/BluetoothPeripheral: disconnected 'BLESmart_0000021FEC21E500D0D6' on request
2022-08-04 15:48:55.085 15479-16038/com.example.bt_wear_spielwiese D/BluetoothGatt: close()
2022-08-04 15:48:55.097 15479-16038/com.example.bt_wear_spielwiese D/BluetoothGatt: unregisterApp() - mClientIf=7
2022-08-04 15:48:55.123 15479-16060/com.example.bt_wear_spielwiese E/BluetoothHandler: Peripheral BLESmart_0000021FEC21E500D0D6 has DISCONNECTED

weliem, 2022-08-04:
No, never tried it on Wear.....

As you can see bonding is failing. My library isn't doing much during the bonding so I have no idea what is going wrong....

vkjambit, 2022-08-05:
yes i thought so. thanks. if i find a solution i will let you know.

closed 2022-08-05 by vkjambit