weliem/blessed-android-coroutines #32: ScanRecord not available in BluetoothPeripheral

nnemanjakg, 2022-09-28:
Hi,

I'm using your library and everything is fine but currently we have a request to check for advertising flags once we get scanning results. Currently method : 
`    private fun sendScanResult(result: ScanResult) {
        scope.launch {
            if (isScanning) {
                val peripheral = getPeripheral(result.device.address)
                peripheral.setDevice(result.device)
                currentResultCallback?.invoke(peripheral, result)
            }
        }
    }`

assigns only device (BluetoothDevice) but there is no support for ScanRecord.
Is this something that is in planned to be supported by the lib, or is there some workaround? 
Are you ok to create a PR to support this?

Best regards,
Nemanja

weliem, 2022-09-29:
The ScanRecord is part of the ScanResult object...

nnemanjakg, 2022-09-29:
I know, that is why I ask if it will be somehow supported by the library as currently I can't get ScanResult as scanning only returns BluetoothPeripheral and not some wrapper class that contains BluetoothPeripheral and for example ScanResult. 
I mean I can implement scan without library and use library for other cases but this seems like a good feature to be added.

Best regards
Nemanja

weliem, 2022-09-30:
I don't really understand your question. As you can see in the code snippet you posted, both a BluetoothPeripheral and a ScanResult are returned while scanning. 

Please check the example app code on how to use the ScanResult.