weliem/blessed-android-coroutines #36: NullPointerException in BluetoothPeripheralManager

cren90, 2023-03-21:
Looks like the BluetoothPeripheralManager does not handle null device names well in the `handleDeviceConnected` function. Looks like the line linked below is the culprit for this case, but I think others will fail similarly.

https://github.com/weliem/blessed-android-coroutines/blob/678b4ebeb249d65382790ba3eb8f0f0aaf132459/blessed/src/main/java/com/welie/blessed/BluetoothPeripheralManager.kt#L91

[image]

cren90, 2023-03-21:
I think for a change, something like 
`Logger.i(TAG, "Central '%s' (%s) connected", device.name ?: "null", device.address)`

I'm happy to put up a PR with the change(s?) but wanted to get thoughts/other options first

weliem, 2023-03-21:
Good catch, I am happy to accept a PR with your suggestion....

cren90, 2023-03-21:
@weliem I've created a branch but I'm unable to push (403 error), looks like I don't have permissions. I've also made the same change for https://github.com/weliem/blessed-android/issues/181 on the non-coroutines repo, with the same issue pushing.

cren90, 2023-03-24:
@weliem Any thoughts?

weliem, 2023-03-26:
Normally with Github you have to make a fork first and then do PR....

anyway, did the fix for you and made a new release: 0.3.3

cren90, 2023-04-21:
Found another instance of this one. Posted https://github.com/weliem/blessed-android-coroutines/pull/38 to fix

closed 2023-08-12 by weliem