weliem/blessed-android-coroutines #40: compose ui

mordeshen, 2023-05-01:
Has Blessed been tested on Compose UI?

weliem, 2023-05-01:
Not sure what you mean. This is a BLE library that has nothing to do with any UI framework....

mordeshen, 2023-05-02:
Thanks, I mean, I tried to work with the library and everything went very well, but when I changed just the UI from view to Compose, the changes of the data from the device(onCharacteristicChanged) didn't receive. tried a couple of ways(lifecycles stuff mainly), and because I didn't really try to dive into the library, I wondered if this strange case, as u said, tested.

weliem, 2023-05-02:
I can't think of a reason why it wouldn't work in Compose. Just make sure you don't mess up the threading part...

mordeshen, 2023-05-02:
thanks. appreciate your work!

closed 2023-08-12 by weliem