weliem/blessed-android-coroutines #43: Unexpected Disconnection Due to Timer

VeMax, 2024-07-26:
Hi,

I’ve been experiencing an issue where the device disconnects shortly after establishing a connection. It seems that the disconnection is triggered by a connection timer, although this shouldn’t be happening. I use the latest 0.4.1 version.

• The connection process begins and successfully reaches the CONNECTED state (I use `centralManager.connectPeripheral()`).
• Shortly after, the startConnectionTimer triggers a disconnection due to a timeout, although the connection should be stable.
• The timer is set to disconnect the peripheral even if the connection was successful.

Here are the logs:
[disconnection-logs.txt](https://github.com/user-attachments/files/16391011/disconnection-logs.txt)

VeMax, 2024-07-26:
I found the problem.

Steps to Reproduce:
	1. Ensure nRF Connect is connected to the target device.
	2. Use the Blessed library to connect to the same device using `central.connectPeripheral(central.getPeripheral("macAddress"))`
	3. Observe the logs for connection and disconnection events.
	
I added logs to find the problem
```kotlin
private fun startConnectionTimer(peripheral: BluetoothPeripheral) {
        cancelConnectionTimer()

        timeoutJob = scope.launch {
            Timber.i("TEST_TEST timeoutJob started") // logs
            delay(CONNECTION_TIMEOUT_IN_MS)
            Logger.e(TAG, "connection timeout, disconnecting '%s'", peripheral.name)
            disconnect()
            scope.launch {
                delay(50)
                bluetoothGatt?.let {
                    bluetoothGattCallback.onConnectionStateChange(bluetoothGatt, HciStatus.CONNECTION_FAILED_ESTABLISHMENT.value, BluetoothProfile.STATE_DISCONNECTED)
                }
            }
        }
    }
``` 
```kotlin
/**
     * This abstract class is used to implement BluetoothGatt callbacks.
     */
    private val bluetoothGattCallback: BluetoothGattCallback = object : BluetoothGattCallback() {
        override fun onConnectionStateChange(gatt: BluetoothGatt, status: Int, newState: Int) {
            Timber.i("TEST_TEST cancelled: ${newState != BluetoothProfile.STATE_CONNECTING} " + // logs
                    "gatt=$gatt status=$status newstate=$newState") // logs
            if (newState != BluetoothProfile.STATE_CONNECTING) cancelConnectionTimer()
            val previousState = state
            state = newState

            val hciStatus = HciStatus.fromValue(status)
            if (hciStatus == HciStatus.SUCCESS) {
                when (newState) {
                    BluetoothProfile.STATE_CONNECTED -> successfullyConnected()
                    BluetoothProfile.STATE_DISCONNECTED -> successfullyDisconnected(previousState)
                    BluetoothProfile.STATE_DISCONNECTING -> {
                        Logger.d(TAG, "peripheral is disconnecting")
                        listener.disconnecting(this@BluetoothPeripheral)
                    }
                    BluetoothProfile.STATE_CONNECTING -> {
                        Logger.d(TAG, "peripheral is connecting")
                        listener.connecting(this@BluetoothPeripheral)
                    }
                    else -> Logger.e(TAG, "unknown state received")
                }
            } else {
                connectionStateChangeUnsuccessful(hciStatus, previousState, newState)
            }
        }
     }
``` 

And here you can see the logs 
[disconnect-with-logs.txt](https://github.com/user-attachments/files/16391996/disconnect-with-logs.txt)

The problem is that you cancel job before it started. By the way, the problem appeared in version 0.3.2
[image]

weliem, 2024-07-27:
Indeed, there is a possible race conditions. Fixed by 67cedddec56269e473adb595ad00b07cc9220ceb

closed 2024-07-27 by weliem