weliem/blessed-android-coroutines #44: Blood Pressure Units Flag calculated incorrectly

imsunburnt, 2024-09-10:
The Blood Pressure Units Flag is calculated incorrectly in file BloodPressureMeasurement.kt.

Suggested fix:

In file com.welie.blessedexample.BloodPressureManagement.kt line 25:
Change unit calculation 
From:

`
  val unit = if (flags and 0x01 > 0) ObservationUnit.MMHG else ObservationUnit.KPA
`

To:

`
  val unit = if (flags and 0x01 > 0) ObservationUnit.KPA else ObservationUnit.MMHG
`

Reason:
The definition of BloodPressureMeasurement characteristic from org.bluetooth.characteristic.blood_pressure_measurement.yaml:

  <...>

  fields:
    - name: Flags
      description: |-
        These flags define which data fields are present in the characteristic value.
        The bits of this field are defined below.
      values:
        - bit: "0"
          description: |-
            Blood Pressure Units Flag
            0 = Blood pressure for Systolic, Diastolic and MAP in units of mmHg
            1 = Blood pressure for Systolic, Diastolic and MAP in units of kPa

  <...>

Verified using Omron Model BP5450.

Might this problem exist in other Blessed repositories?

weliem, 2024-09-10:
Looks like I fixed this in the other Blessed libraries but not in this one......see for example https://github.com/weliem/blessed-kotlin/blob/main/app/src/main/java/com/example/blessed3/BloodPressureMeasurement.kt#L35

Will fix it

weliem, 2024-09-10:
Fixed by [67ceddd](https://github.com/weliem/blessed-android-coroutines/commit/67cedddec56269e473adb595ad00b07cc9220ceb)

closed 2024-09-10 by weliem