weliem/blessed-android-coroutines #46: Question: Why use Executors.newSingleThreadExecutor() for callback tasks?

Ilyas4869, 2024-10-23:
Hello,

First I'd like to thank you for developping this library, it's really helpful !

I'm trying to understand the code of the library so that I rewrite the peripheralManager to use coroutines as well, since I need to use the library on a server device.

My question is why have a specific thread allocated for callbackScope in BluetoothPeripheral.kt instead of just using the IO or Default dispatchers scope? 

Thank you a lot!

weliem, 2024-10-23:
Hi

If you use Dispatchers you are using a thread pool. Hence there is a small risk of callbacks not executing in their sequence of creation. I have had this issue a couple of times in the past. My notifications started to arrive out of order to the app layer. Only occasionally happened but still a problem that caused my app to malfunction.

To be 100% sure that callbacks always are delivered in the right order, I use a single thread executor. But a Dispatcher with only 1 thread combined with a queue can also work.

Ilyas4869, 2024-10-23:
It's very clear. Thank you for the quick answer and the explanation 🙏